enriquewph/Impresora-Braille #7: Cargar comentarios y cosas investigadas/hechas a la wiki.

enriquewph, 2019-06-13:
Jose ponete las pilas capo

enriquewph, 2019-06-14:
@jmora000  lo de caracteristicas del texto en braille esta joya, en como acceder a el diagrama gantt mirate el comentario que te hize en esa tarea.
En la pagina del presupuesto incluir una foto y dejarlo un poco mas lindo xd fijate si podes insertar la tabla como html o imagen que se actualize automaticamente con plugins de google sheets
@maxirivadero deja menos LGBT el doc del google sheets, un poco mas formal pls xd

@jmora000  en diseño y pruebas de un solenoide cambiar el titulo a "Solenoide: La pieza central"
e incluir imagenes extraidas de los modelos de solidworks.

enriquewph, 2019-06-21:
@jmora000 en diseño y pruebas de un solenoide cambiar el titulo a "Solenoide: La pieza central"
e incluir imagenes extraidas de los modelos de solidworks.
En la pagina del presupuesto incluir una foto y dejarlo un poco mas lindo xd fijate si podes insertar la tabla como html o imagen que se actualize automaticamente con plugins de google sheets
@jmora000 incluir la parte mecanica con detalle de sensores y motores y eso.

jmora000, 2019-06-21:
Oye tranquilo viejo

jmora000, 2019-06-21:
[image]

jmora000, 2019-06-21:
[image]

[image]

[image]

[image]

jmora000, 2019-06-21:
@jmora000 incluir la parte mecanica con detalle de sensores y motores y eso.

Y eso? Lucas me dijo que los sensores dijimos que no

enriquewph, 2019-06-21:
> @jmora000 incluir la parte mecanica con detalle de sensores y motores y eso.
> 
> Y eso? Lucas me dijo que los sensores dijimos que no

wut solo inclui mis comentarios en la tarea xd

enriquewph, 2019-09-05:
@jmora000 ahora hay que incluir lo del diseño electronico, con fotos del pcb y esquematico

closed 2019-11-12 by enriquewph