enriquewph/Impresora-Braille #12: desarrollar driver/buscar driver para motor CC de la impresora

enriquewph, 2019-06-14:
Ver que opciones hay para drivers de motores CC. buscar algo que tenga baja resistencia on y alta eficiencia
idealmente que integren mosfets con RDSon <0.5Ohm

El motor lo vamos a usar a 12v con pwm a 8/10/12/15bit segun haga falta jejejejejjej

fijarse x mercadolibre y tratar de ver que es lo que usa la impresora originalmente.

Considerar que la fuente de la impresora es de como 30v (HAY QUE MEDIRLA TODAVIA), verificar corriente de funcionamiento y corriente de stall

EL DRIVER DEBE INCLUIR UN SENSOR DE CORRIENTE PARA DETECTAR CUANDO SE LLEGO AL EXTREMO DE UN EJE PARA SETEAR LA POSICION X=0!

ppnf2, 2019-07-26:
Driver pap A4988
Driver cc L293d

ppnf2, 2019-07-26:
[image]

ppnf2, 2019-07-26:
[image]

enriquewph, 2019-08-02:
se va a ir para el motor CC de la impresora con el integrado DRV8871, en combinacion con un sensor de corriente para sensar cuando el motor se detuvo contra el fin del carrete.
[image]
[drv8871.pdf](https://github.com/enriquewph/Impresora-Braille/files/3460943/drv8871.pdf)

closed 2019-08-02 by enriquewph